MyTooliT/ICOc #26: Readme Update

LoydIrrving, 2022-05-10:
The Readme could maybe need an update about calling icoc with additional parameters like -h , -n , -p , ...
Could be seen as trivial but it wouldn't be bad to implement it in the Readme.

sanssecours, 2022-05-10:
I think the most important command line switches should be already documented [here](https://mytoolit.github.io/ICOc/#command-line-usage-of-icoc).

sanssecours, 2022-05-10:
I just moved most of the information from the readme into a new file in the `Documentation` directory. I also added two addtional links – one right at the start – to the [online documentation](https://mytoolit.github.io/ICOc/) to the readme. This should improve the chance that people discover that more information (than the one in the readme) is available.

In addition to the changes above I also added a link to the [“tutorial” section](https://mytoolit.github.io/ICOc/#tutorials) section at the end of the section [“Basic Usage”](https://mytoolit.github.io/ICOc/#basic-usage). Hopefully this way people find their way to the [documentation of the command line interface](https://mytoolit.github.io/ICOc/#command-line-usage-of-icoc) more easily.

I will close this issue for now. Feel free to reopen it, if you still think that we should extend the documentation about the CLI interface.

closed 2022-05-10 by sanssecours