MyTooliT/ICOc #38: Connection Issues with more than 1 device

tonejca, 2023-02-13:
# Steps to reproduce:
- Execute the mqttronic software and connect to one of devices in close distance (5cm).

# Expected outcome:
- The STH with the specified device name will respond in all cases.

# Actual Outcome:
- Sometimes: refusal of connection + error message: no connection within certain time + suggestion, that another device (the one close by) with ID <some hex ID> would be available.

sanssecours, 2023-02-14:
Thank you for the bug report 💖. Regarding the “actual outcome”: Does the device you wanted to connect to also show up in the available devices or does the output only list the other device?

sanssecours, 2023-02-14:
> Does the device you wanted to connect to also show up in the available devices or does the output only list the other device?

Only the other device did show up.