MyTooliT/ICOc #45: STU test does not work because of report file

bertlderschreckliche, 2023-09-05:
When I try to test the STU by running "test-stu -k eeprom -k connection" I get an error that the report can't be created. 

The production date seems to be problematic as well.

`PS C:\Users\nleder\Documents\STU_flash> test-stu -k connection
.

General
———————
ICOc Version      1.6.0
Date         2023-09-05
Time           15:43:17
Operator     Dummy user

Attributes
——————————
Bluetooth Address 14-2D-41-E5-7F-5A
Firmware Version             2.1.11
Release Name                Valerie
E
======================================================================
ERROR: tearDownClass (mytoolit.test.production.stu.TestSTU)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 164, in apply
    getattr(doc,arn)(*args)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 1293, in handle_pageBegin
    self._handle_pageBegin()
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 620, in handle_pageBegin
    self.pageTemplate.onPage(self.canv,self)
  File "C:\Users\nleder\Documents\GitHub\ICOc\mytoolit\report\report.py", line 39, in _first_page
    PDFImage(logo_filepath, logo_width, logo_height).drawOn(
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nleder\Documents\GitHub\ICOc\mytoolit\report\pdf.py", line 35, in __init__
    with open(filepath, "rb") as pdf_data:
         ^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'mytoolit\\report\\MyTooliT.pdf'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\nleder\Documents\GitHub\ICOc\mytoolit\test\production\node.py", line 124, in tearDownClass
    cls.report.build()
  File "C:\Users\nleder\Documents\GitHub\ICOc\mytoolit\report\report.py", line 250, in build
    self.document.build(self.story, onFirstPage=first_page)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 1317, in build
    BaseDocTemplate.build(self,flowables, canvasmaker=canvasmaker)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 1079, in build
    self.clean_hanging()
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 591, in clean_hanging
    self.handle_flowable(self._hanging)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 925, in handle_flowable
    f.apply(self)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 171, in apply
    annotateException("\nhandle_%s args=%s"%(action,ascii(args)))
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\lib\utils.py", line 1176, in annotateException
    rl_reraise(t,t(sep.join((_ for _ in (msg,str(v),postMsg) if _))),b)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\lib\utils.py", line 138, in rl_reraise
    raise v.with_traceback(b)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 164, in apply
    getattr(doc,arn)(*args)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 1293, in handle_pageBegin
    self._handle_pageBegin()
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 620, in handle_pageBegin
    self.pageTemplate.onPage(self.canv,self)
  File "C:\Users\nleder\Documents\GitHub\ICOc\mytoolit\report\report.py", line 39, in _first_page
    PDFImage(logo_filepath, logo_width, logo_height).drawOn(
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nleder\Documents\GitHub\ICOc\mytoolit\report\pdf.py", line 35, in __init__
    with open(filepath, "rb") as pdf_data:
[... 73 lines cut ...]
    annotateException("\nhandle_%s args=%s"%(action,ascii(args)))
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\lib\utils.py", line 1176, in annotateException
    rl_reraise(t,t(sep.join((_ for _ in (msg,str(v),postMsg) if _))),b)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\lib\utils.py", line 138, in rl_reraise
    raise v.with_traceback(b)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 164, in apply
    getattr(doc,arn)(*args)
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 1293, in handle_pageBegin
    self._handle_pageBegin()
  File "C:\Users\nleder\AppData\Local\Programs\Python\Python311\Lib\site-packages\reportlab\platypus\doctemplate.py", line 620, in handle_pageBegin
    self.pageTemplate.onPage(self.canv,self)
  File "C:\Users\nleder\Documents\GitHub\ICOc\mytoolit\report\report.py", line 39, in _first_page
    PDFImage(logo_filepath, logo_width, logo_height).drawOn(
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nleder\Documents\GitHub\ICOc\mytoolit\report\pdf.py", line 35, in __init__
    with open(filepath, "rb") as pdf_data:
         ^^^^^^^^^^^^^^^^^^^^
FileNotFoundError:
handle_pageBegin args=() [Errno 2] No such file or directory: 'mytoolit\\report\\MyTooliT.pdf'

----------------------------------------------------------------------
Ran 2 tests in 8.355s

FAILED (errors=1)
PS C:\Users\nleder\Documents\STU_flash>`

sanssecours, 2023-09-11:
I looked into this issue again and tested the following commands:

[user configuration]: https://mytoolit.github.io/ICOc/#introduction:section:changing-configuration-values

```sh
cd ~/Downloads
pip uninstall icoc
pip install icoc
test-stu -k eeprom -k connection
```

- once with [user configuration][] and 
- once without user configuration

on

- a Mac and
- a Linux machine.

> The production date seems to be problematic as well.

Sorry for the late response. I just skipped over this text when I first read your problem report. You can change the production date by adding the following text to your [user configuration][]:

```yaml
stu:
  production date: 2023-09-11
```

I added this mapping to the default user configuration in commit 369a0358.

sanssecours, 2023-10-18:
Just remembered another possibility why the test commands do not work on your machine. Maye `pip` cached an old/broken version of the ICOc package on your machine. While this is more or less a wild guess, maybe the following commands:

```sh
pip uninstall icoc
pip install --no-cache-dir icoc
```

might fix your problem.

sanssecours, 2024-06-19:
This should be fixed in the latest version of ICOc (`1.10.0`), at least as far as I can tell from a video conference with @bertlderschreckliche.

closed 2024-06-19 by sanssecours